dhhagan/py-opc #10: PM1 / PM2.5 / PM10 Readings are incorrect (Firmware >= 17)

simonkemper, 2015-10-20:
Using the sensor with a calibrated particle counter reading of PM1 / PM2.5 / PM10 should be in a range of 10-40 µg. Using your library the converted floats in the histogram are somewhat between 2000.00 - 2200.00. Any ideas? Maybe the MSB / LSB are inverted?

dhhagan, 2015-10-20:
Interesting. When you say you are using the converted floats in the histogram, do you mean you are using the values labeled 'pm1', 'pm25', and 'pm10'? Or are you integrating across the bins to get those values? Also, are you performing an instantaneous reading? Or are you averaging/summing across multiple iterations?  I have seven OPC-N2's here that are running great; this is quite curious and I'd love to help!

simonkemper, 2015-10-20:
Yes. I am using those values labeled 'pm1' ... 'pm10'. I am directly reading the output I get when calling the 'read_histogram()' function. I am doing this once every second. I haven't figured out yet what went wrong.

dhhagan, 2015-10-20:
Interesting. What are the values like across the 16 bins?

simonkemper, 2015-10-20:
Well they are somehow strange two. The first to bins (start at bin0) are somewhere between 20-30. Bin4 is always in a range of 1500 - 1600 no matter what you do. The last bins are always 0 and only moving up to 1 - even if applying a test aerosol ...

dhhagan, 2015-10-20:
Interesting. There is clearly something strange going on. Have you tried running it with their software? If so, do you see something similar?

simonkemper, 2015-10-20:
Yes I have done that. And everything is running as expected. I will clean the system and do a reinstall of the whole software - hopefully that will help. I will let you know. - However, I was hoping this might be a known issue and that there might exist a fix. Thanks anyway!

dhhagan, 2015-10-20:
You are using the opc.OPCN2 class, correct?

simonkemper, 2015-10-21:
After doing a clean reinstall of almost everything I found out what caused the bad reading. I somehow installed it to a +3.3V rail. Interesting that communications where still possible. I really had no clue because the everything seem to work - except for the software. 

The Values I get are now much more useful. I will close this for now. Anyway I might still have some questions concerning average calculations, etc. I will use the email that if seen here somewhere! Thanks,

Simon

closed 2015-10-21 by simonkemper