dhhagan/py-opc #14: RPi connection failure - FirmwareError

Myszka, 2016-01-04:
I can not connect with OPC-N2 sensor, which reports under the Alphasense software firmware version: "FirmwareVer=OPC-017b...............................BD".
During initialization with command:
```alpha = opc.OPCN2(spi)```
I get
```FirmwareError: Cannot determine correct firmware for this OPC: ```

Additionally in the official manual I found different wiring for the molex 6w plug for OPC-N2 (attached).
[image]

dhhagan, 2016-01-04:
Yup. You're correct. These are pinouts from a breakout board I had designed for the board. I'm very sorry about this. I will change it in the docs immediately.

louisanb, 2016-03-18:
Sorry i don't get what is wrong with the wiring? I got the same error with 
SCK-CLK
SDO-MISO
SDI-MOSI
/SS-CE0
Is this the right one ?

dhhagan, 2016-03-18:
Can you give me a bit more information? What firmware version is your opc?

louisanb, 2016-03-18:
OPC 1.0.5591.26628
I looked it up in the document it came with and it says firmware can be found when it is first connected to the software and this is what is shown. "0.5591" is the version?

louisanb, 2016-03-18:
oh sorry Firmware version=17
Found USB-ISS versioin 6
USBISS SerialNo=00021780
Device interface=USBISS
Device info String=OPC-N2 FirmwareVer=OPC
Device type set to OPC_N2
Firmware version=17

dhhagan, 2016-03-18:
Ok okay! It should work then. How are you powering it?

louisanb, 2016-03-18:
A battery pack with 4 AA 1.5V i measured the output volts out of the battery pack which is 5.6V

dhhagan, 2016-03-18:
Is it powering both the pi and the opc?

dhhagan, 2016-03-18:
Can you attach a picture or diagram?

louisanb, 2016-03-18:
No just the opc. i am using the wall power cable thing came with the pi for powering the pi alone right now

dhhagan, 2016-03-18:
Ahh okay..there may be some grounding issues but probably not. Do you know
if the batteries supply enough current? It needs at least 260 mA

dhhagan, 2016-03-18:
Otherwise I would try verifying the connections are good?

louisanb, 2016-03-18:
oh i see! it might be the connections which seems solid but it maybe causing the problem here. The ground issues thing is a good call too i will check that. 
About the Current I m using regular Energizer battery [http://www.amazon.com/Energizer-Max-Alkaline-Battery-24-Count/dp/B004U429AQ/ref=sr_1_1_s_it?s=hpc&ie=UTF8&qid=1458342479&sr=1-1&keywords=energizer+batteries](url) 
I assume it is fine the fun will be turned on when i connect the batter pack (it takes about a minute for the fun to be turned on which i assume is an indicator it is being powers properly)and that is how i know the connection is good ( at least the power ones and i assume others should be good too but i will try to find a way to check)

louisanb, 2016-03-18:
I am checking the current now!

dhhagan, 2016-03-21:
Hey @louisanb did you solve your issues with the OPC?

louisanb, 2016-03-24:
Oh sorry i did not see this. Thank you very much for asking! I just got the perfect wire and crimp that i bought to eliminate bad connections concern today. It still has the same error. I am still trying to solve the problem and i am going to do some troubleshooting on the grounding issue you mentions before.  besides that i noticed i have to only connect the power and GND pin to power it up first then connect other pins on Rpi to at least turn OPC on. If i just connect all the pins all at once the OPC won't be turned on. Also when the power source is off the and all connection remain the same, the voltage between SCK-SDI is 2.75v 
SCK-SDI=2.75 SCK-SDO=.41 SCK-SS=.37 SCK-Vcc=1.89 SCK-GND=.77 
SDI-SDO=2.3 SDI-SS=2.3 SDI-Vcc=.86 SDI-GND=1.98 
SDO-SS=.034 SDO-Vcc=1.48 SDO-GND=.36 
SS-Vcc=1.51 SS-GND=.39  
Vcc-GND=1.13 
vs power on: Number in () is the measurement collected once in while which i think is measurement errors
SCK-SDI= 3.3v SCK-SDO=0 SCK-SS=0 SCK-Vcc=2.77 (5.5) SCK-GND=1.18 (0)
SDI-SDO=3.29 SDI-SS=3.29 SDI-Vcc=2.17 (0) SDI-GND=3.3 (1.18 4.5)
SDO-SS=.0 SDO-Vcc=5.5 SDO-GND=0 
SS-Vcc=2.76 SS-GND=0 
Vcc-GND=5.5 
[image]

louisanb, 2016-03-25:
sorry forgot to attack the pictures
[image]
[image]

dhhagan, 2016-03-25:
Awesome. Any chance you could tell me which color wires are where on the OPC?

louisanb, 2016-03-25:
Oh yea 
Vcc-Red
SCK-Brown
SDO-Green
SDI-Yellow (connects to purple wire on Rpi) 
SS-Blue
GND-Black

dhhagan, 2016-03-25:
Okay. I can't see the connection to the OPC which is the most likely error. Can you label which colors are plugged into pins 1-6 on the OPC?

louisanb, 2016-03-25:
1-Vcc-Red (1 being bottom left of OPC as shown in the picture below
2-SCK-Brown
3-SDO-Green
4-SDI-Yellow (connects to purple wire on Rpi) 
5-SS-Blue
6-GND-Black
[image]
[image]
[image]

louisanb, 2016-03-25:
For Rpi to read the firmware of OPC, does OPC need to be powered? Also please discard the first picture set i accidently switched the blue pin/wire with the purple one when i took the picture

dhhagan, 2016-03-25:
Hmm. The wiring is definitely fine. I assume it's a connection? Otherwise I'm not sure. Have you enabled SPI on the RPi through rapsi-config?

louisanb, 2016-03-26:
Yea i did enable it. I did not do much on Rpi codewise besides debugging or trying to make the code/OPC to work. The only thing i did was enable SPI and download the code/library from your github and tried to run it and make it work with OPC. I did not check the SPI alone after enable it like the way in one of the pictures below. 
[image]
[image]
[image]

dhhagan, 2016-03-26:
@louisanb Interesting. I'm not exactly sure what's going on then aside from it being a bad connection. If you still are powering them separately, I would try using a central PSU.

louisanb, 2016-03-29:
Thank you very much! I will definitely try the PSU, i feel like it is the cause.

dhhagan, 2016-11-17:
Hey @louisanb any update on this?

louisanb, 2016-11-23:
Hi David:

  I don't work there anymore and thank you very much for checking!

louis

closed 2016-11-23 by dhhagan