dhhagan/py-opc #15: Using the library for accessing a device connected via USB

rouvoy, 2016-02-08:
My OPC-N2 device is connected to my Raspberry Pi 2 using a USB adapter, and the current version of the library fails to connect to the device. Did you give a try to such settings?

dhhagan, 2016-02-08:
That is correct. I assume you are using a SPI-USB adapter? The library is written to communicate directly over the Raspberry Pi's SPI interface only. You will need to connect the Pi to the OPC-N2 as shown in the documents, or write SPI-USB drivers for the original Alphasense code. Let me know if you have any more questions.

closed 2016-02-12 by dhhagan

MarceloYung, 2016-09-24:
How can I write SPI-USB drivers for the original Alphasense code?

DancingQuanta, 2016-09-24:
What SPI-USB devices are you trying to use with alphasense?

dhhagan, 2016-09-24:
@MarceloYung And what do you mean by the 'original Alphasense code"? The only software they offer is a windows-based GUI; is this what you're referring to?

MarceloYung, 2016-09-24:
I am using the py-opc library for an OPC-N2 sensor, and it doesn't work. And I am connecting the raspberry PI 2 and the OPC-N2 with a SPI-USB adapter. The connection doesn't start using the library.
I tried the sensor in Windows application (provided by Alphasense) and it worked correctly.
I was reading this page and you said that the OPC-N2 and that the library is written to communicate directly over the Raspberry Pi's SPI interface only, and I am asking how to comunnicate when I use an USB adapter, as I show in the attached image file.
The version of my OPC-N2 is v18.2. 
Thanks for your help!!
[image]

DancingQuanta, 2016-09-24:
Ah, I wrote such a driver for that adapter, that adapter is called USB-ISS.
My driver is a python module at https://github.com/DancingQuanta/pyusbiss.
You have to download it and install using the instructions on my README.
Please open an issue over there if you have any issues.
I also have an example script that I will upload soon.

DancingQuanta, 2016-09-24:
@MarceloYung I have uploaded the example script here https://github.com/DancingQuanta/pyusbopc.
Please open an issue over there if you have any problems

MarceloYung, 2016-09-24:
Thanks very much, my project is working!!

DancingQuanta, 2016-09-24:
Fantastic