dhhagan/py-opc #30: Add support for Firmware v17b

rouvoy, 2016-07-28:
Does anyone succeed in running py-opc on a OPC-N2 v.17b?

I got the error `Your firmware version could not be automatically detected.`, so far...

dhhagan, 2016-07-28:
Yes, it should. I used to run it on 17.b. Typically, it means there is a bad connection..? How do you have it wired up?

rouvoy, 2016-07-28:
I use a separate PSU (5V, 1000mA) to power the OPC-N2 and plug the other cables as documented:
- `SCK =>  CLK`
- `SDO =>MISO`
- `SDI => MOSI`
- `/SS => CE0`
- `GND => GND`

[image]

dhhagan, 2016-07-28:
Interesting. I will try to find an OPC w/fW 17.b and test myself. You're using the most up-to-date version of the library, I assume? What version of python are you using?

Can you try running the following:

    import spidev
    import time

    spi = spidev.SpiDev(0, 0)
    spi.open(0, 0)
    spi.mode = 1
    spi.max_speed_hz = 500000

    info_string = []
    spi.xfer([0x3F])
    time.sleep(9e-3)
    
    for i in range(60):
        resp = spi.xfer([0x00])
        info_string.append(chr(resp))

    ''.join(info_string)

rouvoy, 2016-07-29:
I use Python 2.7.9 with the most up-to-date version of your library.

The above code display nothing in the console.

dhhagan, 2016-07-29:
Okay. Is the OPC starting up after about a minute even without sending a command? Have you run any other SPI devices off the RPi? It seems like one of the pins has a poor connection (MOSI, MISO, or CS).

What does `lsmod` show?

rouvoy, 2016-07-29:
1. The OPC starts automatically after a minute without sending any command
2. I never run any other SPI device on the RPi2
3. Here is the result of `lsmod`:
```
Module                  Size  Used by
cfg80211              427855  0 
rfkill                 16037  1 cfg80211
ftdi_sio               31193  0 
usbserial              22115  1 ftdi_sio
evdev                  11396  3 
snd_bcm2835            20511  0 
snd_pcm                75698  1 snd_bcm2835
snd_timer              19160  1 snd_pcm
snd                    51844  3 snd_bcm2835,snd_timer,snd_pcm
bcm2835_gpiomem         3040  0 
spi_bcm2835             7286  0 
bcm2835_wdt             3225  0 
i2c_bcm2708             4770  0 
uio_pdrv_genirq         3164  0 
hid_apple               4925  0 
uio                     8000  1 uio_pdrv_genirq
i2c_dev                 5859  0 
fuse                   83461  1 
ipv6                  347530  30 
```

dhhagan, 2016-09-02:
Sorry this has taken so long; I am finally back in the US.

Can you tell me what the command `pip freeze` prints out?

gvoelker, 2016-09-15:
I have excat the same issue with the difference that my OPC-N2 is running on firmeware version 18. Raspi and OPC are powered by the same PSU (max 3A). Looking forward for your answer. Greetings from Germany .

dhhagan, 2016-09-15:
Hi @gvoelker Have you tried doing the steps I outlined above?

Try running this and let me know what you get:

import spidev
import time

```
spi = spidev.SpiDev(0, 0)
spi.open(0, 0)
spi.mode = 1
spi.max_speed_hz = 500000

info_string = []
spi.xfer([0x3F])
time.sleep(9e-3)

for i in range(60):
    resp = spi.xfer([0x00])
    info_string.append(chr(resp))

''.join(info_string)
```

gvoelker, 2016-09-15:
Result of pip freeze:
pi@raspberrypi:~ $ pip freeze
Pillow==2.6.1
RPi.GPIO==0.6.2
RTIMULib==7.2.1
argparse==1.2.1
chardet==2.3.0
colorama==0.3.2
gpiozero==1.2.0
html5lib==0.999
lxkeymap==0.1
mcpi==0.1.1
ndg-httpsclient==0.3.2
numpy==1.8.2
picamera==1.11
pifacecommon==4.2.1
pifacedigitalio==3.1.0
pigpio==1.30
py-opc==1.0.1
pyOpenSSL==0.13.1
pyasn1==0.1.7
pygame==1.9.2a0
pygobject==3.14.0
pyserial==2.6
python-apt==0.9.3.12
requests==2.4.3
sense-hat==2.1.0
six==1.8.0
spidev==3.0
urllib3==1.9.1
wheel==0.24.0
wsgiref==0.1.2
Result of lsmod:
pi@raspberrypi:~ $ lsmod
Module                  Size  Used by
bnep                   10340  2 
hci_uart               17943  1 
btbcm                   5929  1 hci_uart
bluetooth             326105  22 bnep,btbcm,hci_uart
hid_logitech_hidpp     10382  0 
brcmfmac              186343  0 
joydev                  9024  0 
evdev                  11396  6 
brcmutil                5661  1 brcmfmac
cfg80211              427855  1 brcmfmac
hid_logitech_dj        10016  0 
rfkill                 16037  4 cfg80211,bluetooth
snd_bcm2835            20511  1 
snd_pcm                75698  1 snd_bcm2835
snd_timer              19160  1 snd_pcm
snd                    51844  5 snd_bcm2835,snd_timer,snd_pcm
i2c_bcm2708             4770  0 
spi_bcm2835             7286  0 
bcm2835_gpiomem         3040  0 
bcm2835_wdt             3225  0 
uio_pdrv_genirq         3164  0 
uio                     8000  1 uio_pdrv_genirq
i2c_dev                 5859  0 
fuse                   83461  3 
ipv6                  347530  52 
Result of the code:
Traceback (most recent call last):
  File "/home/pi/Documents/OPC-N2/Programme/error_search_1.py", line 14, in <module>
    info_string.append(chr(resp))
TypeError: an integer is required

dhhagan, 2016-09-15:
@gvoelker Can you make sure that MOSI and MISO are hooked up correctly and not switched? Also, try editing the script above to just print out the `resp` to see why it's not giving an integer?

gvoelker, 2016-09-20:
I checked over the the right wiring an made sure tht MOSI and MISO are hooked up correctly. After editing and running the script, the result was \x00.

dhhagan, 2016-09-21:
@gvoelker What version of Python are you using?

gvoelker, 2016-09-22:
i am using Version 2.7.3

DancingQuanta, 2016-09-22:
@gvoelker can you pass `resp` to `type()` like this
```
print(type(resp))
```
to find out what it is.
I wonder whether it is a problem with spidev and python 2 or 3?

dhhagan, 2016-09-22:
It shouldn't be, but still worth a try! I've tested it on both 2.7 and 3.5
and it works great..

gvoelker, 2016-10-19:
I am glad to tell you that almost everything is working fine at the moment. The soultion was to use two different PSUs. One for the Raspi and one for the OPC-N2. I don#t know exactly why that error occurs. One last question, even if it is of this topic: How can in Change the sampling rate and what does 'OPCN2' object has no attribute 'read_Histogram'.

dhhagan, 2016-10-19:
@gvoelker Great! Glad you got it working. To change the sampling rate, you simply sample at whatever rate you would like. Essentially, the histogram will continue to increment until you read the histogram again. So if you want a 1 Hz measurement, read the histogram every second. If you want 0.1 Hz data, read the histogram every 10 seconds. For reference, the Alphasense software by default samples once every 1.4 seconds I believe.

As for the second part, it sounds like you are using the new version of the library >1.0 which had some non back-compatible API changes. You can check the new docs for the version you are using, but `read_histogram()` was changed to `histogram()`. I will be making some more changes within the next couple weeks after I return from a conference, but everything should still work.

closed 2016-10-24 by dhhagan