dhhagan/py-opc #31: Allow choice of usb serial module alongside spidev

DancingQuanta, 2016-09-01:
In my current project I have a sensor connected to SPI to USB converter which I can query using serial module.
I think serial module must be added and I could do a pull request but this involves large reorganisation of code and so I think we need to discuss this.

dhhagan, 2016-09-01:
Hi @DancingQuanta : So if I understand correctly, you have successfully been able to use this library with pyserial and a usb-spi converter rather than GPIO? Have you tested it on anything other than an RPI? I have no problems adding it to the module, but would need to make sure it works on various architectures first. One of the perks of spidev is it works on non-RPi platforms. Feel free to fork it and send a PR though.

DancingQuanta, 2016-09-01:
Hello @dhhagan I spoke too soon. I did not understand how my USB-SPI converter work and now discovered the converter is multi-purpose converter (I2C SPi OI, etc) which means that it requires setting up before tranmsitting any data to sensor and receiving from the sensor.
Now having understood my converter I figured it is much better to integrate this into spidev rather than the py-opc.
Sorry about jumping the gun!
Closing the issue

closed 2016-09-02 by DancingQuanta