dhhagan/py-opc #35: Should bin boundaries be converted to um from ADC by user or not

DancingQuanta, 2016-09-12:
Should the conversion of bin boundary ADC value to um be managed by read_info_string function or by the user?

dhhagan, 2016-09-12:
I would argue for by the user. There are multiple adc values that correspond to the same diameter, so it is more accurate and reliable to leave the conversion up to the user if repeatability is desired.

DancingQuanta, 2016-09-12:
Make sense, there should be a note about that in the docs

DancingQuanta, 2016-09-12:
Out of curiosity, what ADC values for each bin do you normally get (if you already gotten them out of your sensor)?

dhhagan, 2016-09-12:
I can throw up a list of the default values later...the first bin is 0.38 - 0.54 microns (off the top of my head). You can look them up using the [lookup_bin_boundary][1] method. Also, the lookup table can be found [here][2]. The ADC value auto-increments.

I docs will be updated with release v1.2. Sorry about the delays..

[1]: https://github.com/dhhagan/py-opc/blob/master/opc/__init__.py#L157
[2]: https://github.com/dhhagan/py-opc/blob/master/opc/lookup_table.py

DancingQuanta, 2016-09-12:
Not a problem, I already got my bin boundaries though the first bin is 0.54
while I thought it was 0.38.

<https://github.com/dhhagan/py-opc/issues/35#issuecomment-246369217>, or mute
the thread

dhhagan, 2016-09-12:
That's what I get to. I think that it's the value for the right-side (upper end) of bin 0. Bin 0 always starts at 0.38 um and the last bin always ends at 17.5 um no matter what.

DancingQuanta, 2016-09-12:
I find it hard to find any description of the bin boundaries so that why I got a bit ocnfused.
Here is my bin boundaries
```
0.54  0.78  1.05  1.34  1.59  2.07  3.0  4.0  5.0  6.5  8.0  10.0  12.0  14.0  16.0
```
Do those values make sense to you? Are they right of bins?

dhhagan, 2016-09-12:
Yea. Alphasense hasn't been great at documenting this. There are a total of 16 bins. The bin boundaries that are accessible via the configuration variables describe essentially the inner 15 boundaries with the overall min and max left out.

So, Bin 0 is 0.38-0.54 microns and Bin 15 is 16.0-17.5 microns. Does that make sense?

DancingQuanta, 2016-09-12:
Yes it makes sense now, I admit I didn't check the number of bin boundaries.

Will close this issue if there is nothing else to discuss

closed 2016-09-12 by DancingQuanta