dhhagan/py-opc #38: connection on raspberry pi 2 failed.

Jostar1024, 2016-11-17:
I am using the Alphasense OPC-N2 on the raspberry pi 2, but it give me this Startup Error

> Your fireware version could not be automatically detected. This is usually caused by a bad wiring or poor power supply. If neither of these are likely candidates, please open an issue on the Github repo.

how can I fix it ?

This is how the sensor is connected.
[image]

dhhagan, 2016-11-17:
Hi @Jostar1024 : What version of the library are you using? What version of Python are you using? And which Pi is that? Looks like a Pi2?

Also, if you could draw out the wiring diagram, that would help. This error is almost always due to a bad connection, which can be hard to debug at times.

Do you have the USB-SPI connector by any chance?

Jostar1024, 2016-11-17:
Hi,
* Version of library : py-opc(1.1.0)
* Version of python : 2.7.9
* In fact I've tried at both Pi2 and Pi3. In the photo I connect the sensor with the Pi3.

If I fully understand the instruction, here is how I wire.
[image]

I don't have the USB-SPI... Would you tell me how that can help ?

Well I've made every effort to try to wire it correctly... Is it possible that SPI is not working because of the version of the Linux Kernel ?

I've read it at :[I2C, SPI, I2S, LIRC, PPS, stopped working? Read this.
](https://www.raspberrypi.org/forums/viewtopic.php?t=97314)
But the solution mentioned in the article above doesn't work for me...

DancingQuanta, 2016-11-17:
Did you check the voltage level of each connection? Especially power. The
voltage levels at the GPIO pins should be near 0V or 3.3V.
Have you tried SPI loopback test?

Jostar1024, 2016-11-24:
I've checked the voltage level. 
CS0 3.3V the others are around 1V...
I am using a separate power(5V, 1000mA) for the sensor.
Yes I've tried the SPI loopback test, which give me the 00 00 00 ....

patricktokeeffe, 2017-02-21:
I see you are separately powering the OPC using a 5V/1A adapter. In my experience, even slight voltage offsets between OPC ground and RPi ground will result in garbled SPI communications. Try powering the OPC directly from the GPIO rail (5V/G) and ensure you have a quality power adapter (5V @ 1.5 to 2A).

beleme, 2017-06-02:
@Jostar1024  hello 
how can you solved this problem?

dhhagan, 2017-07-03:
@Jostar1024 Did you ever solve this issue?

Jostar1024, 2017-07-04:
I didn't really solve the issue and I used the USB port instead. The time and resources were limited and I'm not working on it anymore.

closed 2017-07-04 by Jostar1024