dhhagan/py-opc #41: opc.exceptions.FirmwareVersionError

agent2003, 2017-01-18:
Hi,

Similar issue to others with the firmware error. I'm using a devantec USB to SPI adapter wired correctly as the windows alphasense tool reports the below and reads the sensor. I therefore don't think its an SPI wiring issue.

Firmware version reported as 18.1 in the windows tool. I note that 18.2 has been tested and works ok.
Found USB-ISS Version 7
USBISS SerialNo = 00030397
Device interface = USBISS
Device Info String = OPC-N2 FirmwareVer=OPC-018.1..............................BD
Device type set to OPC_N2
Firmware version = 18

I'm powering the OPC from a separate USB port on the same power supply as the PI with the grounds connected to avoid issues. I therefore don't think its a power issue.

I've installed pyusbiss and i'm using python 3.4 in the mode where you can type in commands line by line.  When trying the sample code on the website and I run the "alpha = opc.OPCN2(spi)" command (without error catch code), I get the below result. It reads the USB adapter OK but can't connect to the OPC. Any advice would be appreciated.

>>> spi = usbiss.USBISS("/dev/ttyACM1", 'spi', spi_mode = 2, freq = 500000)
The module ID is 7
The firmware version is 0x7
The current operating mode is 0x40
The serial number is b'00030397'
Initializing USB-ISS in SPI mode with 146 spi_mode and 11 sck_divisor
>>> alpha = opc.OPCN2(spi)
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/opc/__init__.py", line 58, in __init__
    self.firmware['version']    = int(re.findall("\d{3}", self.read_info_string())[-1])
  File "/usr/local/lib/python3.4/dist-packages/opc/__init__.py", line 199, in read_info_string
    self.cnxn.xfer([0x3F])
  File "/usr/local/lib/python3.4/dist-packages/pyusbiss-0.1.0-py3.4.egg/usbiss/__init__.py", line 147, in xfer
    decoded = [struct.unpack('B', response[i+1])[0] for i in range(0, len(data))]
  File "/usr/local/lib/python3.4/dist-packages/pyusbiss-0.1.0-py3.4.egg/usbiss/__init__.py", line 147, in <listcomp>
    decoded = [struct.unpack('B', response[i+1])[0] for i in range(0, len(data))]
TypeError: 'int' does not support the buffer interface

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/opc/__init__.py", line 62, in __init__
    self.firmware['version'] = int(re.findall("\d{1}", self.read_info_string())[-1])
  File "/usr/local/lib/python3.4/dist-packages/opc/__init__.py", line 199, in read_info_string
    self.cnxn.xfer([0x3F])
  File "/usr/local/lib/python3.4/dist-packages/pyusbiss-0.1.0-py3.4.egg/usbiss/__init__.py", line 147, in xfer
    decoded = [struct.unpack('B', response[i+1])[0] for i in range(0, len(data))]
  File "/usr/local/lib/python3.4/dist-packages/pyusbiss-0.1.0-py3.4.egg/usbiss/__init__.py", line 147, in <listcomp>
    decoded = [struct.unpack('B', response[i+1])[0] for i in range(0, len(data))]
TypeError: 'int' does not support the buffer interface

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.4/dist-packages/opc/__init__.py", line 240, in __init__
    super(OPCN2, self).__init__(spi_connection, model = 'N2', **kwargs)
  File "/usr/local/lib/python3.4/dist-packages/opc/__init__.py", line 69, in __init__
    raise FirmwareVersionError(msg)
opc.exceptions.FirmwareVersionError: 
                        Your firmware version could not be automatically detected. This is usually caused
                        by a bad wiring or poor power supply. If niether of these are likely candidates, please
                        open an issue on the GitHub repository at https://github.com/dhhagan/py-opc/issues/new

DancingQuanta, 2017-01-18:
Ah I am the author of pyusbiss and I need to add a warning about the startup of OPCN2 with pyusbiss.
In my experience, I get fireware error the first time after powering on the OPCN2 and I have to try two or three times before OPCN2 starts to communciate with me.

Can you repeat `alpha = opc.OPCN2(spi)` up to three times with a minute in the between please?

agent2003, 2017-01-19:
Hi, good work! Thanks for your reply

I've got two OPC-N2 sensors. I've tried what you ask with both of them 5 times with a minute in between and get the same result. Both work fine with the windows tool and the adapter.

Interestingly, when I plug one in, the fan starts up automatically after a while, on the other it does not. I've tried polling it before the fan comes on and after, with the same result.

Do you know of any SPI test commands I can send to try and diagnose the problem? Could it be that firmware version 18.1 is not listed or supported in py-opc?

Thanks

dhhagan, 2017-01-19:
@agent2003 That shouldn't change anything, although I wasn't aware they released publicly a v18.1!  From your traceback above, it looks like it's a simple byte encoding error associated with python3. Can you try running it with Python2 to see if you get the same error? I think I experienced this a while back but fixed it in the pyusbiss library. I sent a [pull request] a while back. @DancingQuanta any updates on this?

[1] https://github.com/DancingQuanta/pyusbiss/pull/2

agent2003, 2017-01-19:
Thanks for your reply. It worked! I changed the python version to 2.7.9, installed the packages again and now it works (after the second try of alpha = opc.OPCN2(spi)). In case it helps anybody, there is a good guide to changing python version here: [https://linuxconfig.org/how-to-change-from-default-to-alternative-python-version-on-debian-linux](url)

Results below:

pi@raspberrypi:~ $ python
Python 2.7.9 (default, Sep 17 2016, 20:26:04) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import usbiss
>>> import opc
>>> spi = usbiss.USBISS("/dev/ttyACM0", 'spi', spi_mode = 2, freq = 500000)
The module ID is 7
The firmware version is 0x7
The current operating mode is 0x40
The serial number is 00030397
Initializing USB-ISS in SPI mode with 146 spi_mode and 11 sck_divisor
>>> alpha = opc.OPCN2(spi)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/opc/__init__.py", line 240, in __init__
    super(OPCN2, self).__init__(spi_connection, model = 'N2', **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/opc/__init__.py", line 69, in __init__
    raise FirmwareVersionError(msg)
opc.exceptions.FirmwareVersionError: 
                        Your firmware version could not be automatically detected. This is usually caused
                        by a bad wiring or poor power supply. If niether of these are likely candidates, please
                        open an issue on the GitHub repository at https://github.com/dhhagan/py-opc/issues/new
                        
>>> alpha = opc.OPCN2(spi)
>>> alpha.on()
True
>>> for key, value in alpha.histogram().items():
...     print ("Key: {}\tValue: {}".format(key, value))
... 
Key: Temperature	Value: None
Key: Bin 10	Value: 0.0
Key: SFR	Value: 3.70000004768
Key: Bin 12	Value: 0.0
Key: Bin 13	Value: 0.0
Key: Bin 14	Value: 0.0
Key: Bin 15	Value: 0.0
Key: PM2.5	Value: 0.0
Key: Bin1 MToF	Value: 0.0
Key: Pressure	Value: None
Key: Bin 11	Value: 0.0
Key: Bin5 MToF	Value: 0.0
Key: PM1	Value: 0.0
Key: Bin7 MToF	Value: 0.0
Key: Checksum	Value: 0
Key: Bin3 MToF	Value: 0.0
Key: PM10	Value: 0.0
Key: Bin 8	Value: 0.0
Key: Bin 9	Value: 0.0
Key: Bin 6	Value: 0.0
Key: Bin 7	Value: 0.0
Key: Bin 4	Value: 0.0
Key: Bin 5	Value: 0.0
Key: Bin 2	Value: 0.0
Key: Bin 3	Value: 0.0
Key: Bin 0	Value: 0.0
Key: Bin 1	Value: 0.0
Key: Sampling Period	Value: 66.3592071533
>>> alpha.off()
True

DancingQuanta, 2017-01-20:
Sorry forgot about that pull request will try to merge today.

DancingQuanta, 2017-01-20:
Merged the [pull request](https://github.com/DancingQuanta/pyusbiss/pull/2).
@agent2003 could you please try out new `pyusbiss` HEAD with python 3?

agent2003, 2017-01-20:
Yeah sure, I will try it this weekend.

closed 2017-01-20 by dhhagan

H60gh, 2018-01-19:
I connect a PM2.5 sensor into Reaspberry pi using SPI. I have followed the example in Python 3.4.2, but it seems "alpha" variable does not work and alpha.ping() returns "False". As I test, spidev works well.
I have switched into Python 2.7.9 ("alpha = opc.OPCN2(spi)" is modified to "alpha = opc.OPC(spi))" and now alpha is unknown variable.
How can I  solve this problem?