dhhagan/py-opc #44: connection error

beleme, 2017-06-02:
Why do I get this type of error?

Startup Error: 'module' object has no attribute 'OPCN2'

dhhagan, 2017-06-02:
@beleme 

What does your code look like?

beleme, 2017-06-02:
I was able to solve that problem last but now I have this problem

this is my code:

import spidev
import opc

spi = spidev.SpiDev()
spi.open(0, 0)
spi.mode = 1
spi.max_speed_hz = 500000

try:
    alpha = opc.OPCN2(spi)
except Exception as e:
    print ("Startup Error: {}".format(e))
else:
    alpha.on()
    print alpha.histogram()
    alpha.off()

and I get:

pi@raspberrypi:~ $ python uno.py 
Startup Error: 
                        Your firmware version could not be automatically detected. This is usually caused
                        by a bad wiring or poor power supply. If niether of these are likely candidates, please
                        open an issue on the GitHub repository at https://github.com/dhhagan/py-opc/issues/new

how can I resolve it?

beleme, 2017-06-02:
@dhhagan

beleme, 2017-06-05:
hello @dhhagan  you can helpme with my problem?

dhhagan, 2017-06-05:
@beleme Have you checked the details on the other issue? The most likely cause is a power issue. It's really hard for me to debug it from afar without knowing exactly how you have it wired, what PSU you are using, etc. Can you tell me more about your setup?

beleme, 2017-06-05:
@dhhagan  ok thanks!

I am connected as is in the py-opc documentation

[image]
as is in the picture

dhhagan, 2017-06-05:
@beleme It looks like you're trying to power the OPC-N2 through the GPIO 5V Pin of the RPi..? I don't think that will provide enough current (needs >250 mA); does the OPC-N2 fan ever start?

beleme, 2017-06-05:
@dhhagan  Yes i am connected the opc with the 5V of the RPi but the fan turns on with that voltage.

The firmware version of OPC-N2 is 18
The version of python is 2.7

dhhagan, 2017-06-05:
Unless you have access to a Logic Analyzer, I would probably focus on trying to get a better voltage source. As mentioned in the docs, you should have a single power source that powers both the Pi and the OPC (not through the GPIO pins).

beleme, 2017-06-06:
@dhhagan  ok

have i to connect the power of the Pi with the micro-usb? or with the port usb?

dhhagan, 2017-06-06:
@beleme It shouldn't matter, but I have used to microUSB

beleme, 2017-06-06:
@dhhagan @DancingQuanta 
I already connected the OPC and the Pi with power supply and it still gives the same problem. The power supply have 5V and 2A.

How can i solve it?

dhhagan, 2017-06-06:
Can you please attach an image or diagram?

beleme, 2017-06-07:
@dhhagan 

[image]
[image]

dhhagan, 2017-06-07:
@beleme  Ahh. I didn't realize you were using the breakout from the SPI-USB converter. Although it's hard to tell from the image, I imagine there is a mixed up pin; the SPI-USB converter Alphasense sells is not ordered (i.e. the pin ordering on the micro-clasp side is different than the larger Molex connector). I would follow those back carefully to make sure they are correctly ordered. 

The other option is to just directly use the RPi with the SPI-USB converter as shown [here][1].
 and [here][2]. 

[1]: http://py-opc.readthedocs.io/en/latest/#using-the-pyusbiss-library-via-usb-port
[2]: http://py-opc.readthedocs.io/en/latest/#connecting-via-a-usb-spi-converter

beleme, 2017-06-07:
already i checked, and unfortunately i committed a mistake in the connection with the terminals of OPC  but now i have this error 

Startup Error: Your firmware is not yet supported. Only versions 14-18 are currently supported. 

my OPC-N2 have a firmware version  is 18

beleme, 2017-06-07:
@dhhagan already I checked, and unfortunately I committed a mistake in the connection with the terminals of OPC  but now I have this error 

Startup Error: Your firmware is not yet supported. Only versions 14-18 are currently supported. 

my OPC have a firmware version 18

dhhagan, 2017-06-07:
@beleme Okay. Your firmware version probably has a minor version as well. Can you let me know what that is?

beleme, 2017-06-07:
@dhhagan    I keep on with the same problem, I will try with the usb converter

JananiVJ, 2017-06-08:
@beleme 
Try this code from @lionfish0 for start up.
import spidev
import opc
from time import sleep
# Open a SPI connection on CE0
spi = spidev.SpiDev()
spi.open(0, 0)

# Set the SPI mode and clock speed
spi.mode = 0b01
spi.max_speed_hz = 10000

#get it used to sending things!  
to_send = [0x3F]
spi.xfer(to_send);

#wait a bit.       
sleep(1.0)
try:
    alpha = opc.OPCN2(spi)
except Exception as e:
    print ("Startup Error: {}".format(e))
print alpha

print alpha.on()
sleep(1.0) #give it time to turn on...

    # Read the histogram and print to console
print alpha.histogram()

semajoel, 2017-06-20:
@JananiVJ  Am new to working with the pie, I have. A pie zero, have setup all correct , have written the code but I get error "Missing parentheses in call to print"

semajoel, 2017-06-20:
@JananiVJ  Am new to the pi, I have all setup including the above code, I get importEr: No module named opc.
How can I install this module?

dhhagan, 2017-06-20:
@semajoel That means you haven't installed the py-opc library. If you go to your terminal and type `pip list`, is opc vx.x listed?

semajoel, 2017-06-21:
@dhhagan i have this listed "py-opc (1.3.0)"

DancingQuanta, 2017-06-21:
@semajoel missing parentheses could mean that you are using Python 3 not Python 2

dhhagan, 2017-06-21:
@semajoel Interesting. Could you post your code along with the version of python you are using?

semajoel, 2017-06-21:
@dhhagan and the rest, I appreciate your efforts,  i figured something errors out, now I have another
spi.open(0,0)
IOError: no such file or directory

DancingQuanta, 2017-06-21:
What is the traceback?

semajoel, 2017-06-21:
Yes, have enabled SPI in pi configuration, every seems fine for now. I can read from the histogram. I have pm data. Thanks a million.

JananiVJ, 2017-06-22:
@DancingQuanta I am using Python 3. and it works for me.

JananiVJ, 2017-06-22:
@dhhagan do you have experience working with any other alphasense sensors? the Nitrogen and ozone sensors to be specific.

dhhagan, 2017-06-22:
@JananiVJ I do, yes. You can email me at dhagan@mit.edu for more details/conversation on that topic.

dhhagan, 2017-07-03:
@semajoel So can you confirm what the error was? I'd like to add it to a FAQ since many people have had similar issues.

semajoel, 2017-07-04:
@dhhagan SPI was not enabled in the configuration of the PI.

closed 2017-07-04 by dhhagan