dhhagan/py-opc #45: continuous operation

JananiVJ, 2017-06-08:
Hello dhhagen
Thank you for the documentations and library.
I am completely new to this and would like to know if anyone managed to get a continuous data output from the OPC.N2 at regular time intervals.
Thank you.

dhhagan, 2017-06-08:
Hey @JananiVJ Yup. I ran 10 of them for 6 months continuously..just make sure your sampling interval >= 1.2 s otherwise it doesn't behave very well.

JananiVJ, 2017-06-08:
When I run the program, the OPC starts and runs continuously. And I get an output in the terminal which looks something like this:

pi@PC-17-0302:~/Documents/examples/opcn2 $ python py-opc.py 
Alphasense OPC-N2v18.2
True
{'Temperature': None, 'Bin 10': 0.0, 'SFR': 3.700000047683716, 'Bin 12': 0.0, 'Bin 13': 0.0, 'Bin 14': 0.0, 'Bin 15': 0.0, 'PM2.5': 0.0, 'Bin1 MToF': 0.0, 'Pressure': None, 'Bin 11': 0.0, 'Bin5 MToF': 0.0, 'PM1': 0.0, 'Bin7 MToF': 0.0, 'Checksum': 0, 'Bin3 MToF': 0.0, 'PM10': 0.0, 'Bin 8': 0.0, 'Bin 9': 0.0, 'Bin 6': 0.0, 'Bin 7': 0.0, 'Bin 4': 0.0, 'Bin 5': 0.0, 'Bin 2': 0.0, 'Bin 3': 0.0, 'Bin 0': 0.0, 'Bin 1': 0.0, 'Sampling Period': 2.7139840126037598}
pi@PC-17-0302:~/Documents/examples/opcn2 $ 

PM values are zero. If I run it again then I get some numbers. How did you manage to get the output data continuously?

dhhagan, 2017-06-08:
@JananiVJ It looks like your sampling period is a bit over 2.5 seconds; is that correct? Where are you operating the device?

JananiVJ, 2017-06-08:
right now for testing, inside a room. In the future I will install these devices in a fixed sites (outdoor). I will also use them in mobile measurements on vehicles and tethered balloons.

DancingQuanta, 2017-06-08:
Could you show your script that runs the OPC?

JananiVJ, 2017-06-27:
@DancingQuanta : This is the script that I use to run the OPC

# Import the OPCN2 module.
import spidev
import opc
from time import sleep

# Open a SPI connection on CE0
spi = spidev.SpiDev()
spi.open(0, 0)

# Set the SPI mode and clock speed
spi.mode = 0b01
spi.max_speed_hz = 10000

# Send OPc data to SPI interface  
to_send = [0x3F]
spi.xfer(to_send);

#wait a bit.       
sleep(1.0)

# Print PM (OPC-N2) sensor values
    try:
        alpha = opc.OPCN2(spi)
    except Exception as e:
        print ("Startup Error: {}".format(e))
    print alpha

    print alpha.on()
    sleep(1.0) #give it time to turn on...
    
    # Read the histogram and print to console
    OPC= alpha.histogram()
    print (OPC)

Note: 
in /usr/local/lib/python2.7/dist-packages/opc/_init_.py

line 76: I set the major and minor versions manually 
line 243: firmware_max = 18.2   # Maximum firmware version supported

closed 2017-06-27 by JananiVJ