dhhagan/py-opc #48: list index out of range

jeffisfast, 2017-06-30:
Thanks for this library.  Trying to use it with a brand new OPC-N2 that I just received and here is what I'm getting:

#python3 pm.py
The module ID is 7
The current operating mode is 0x40
The serial number is b'0034628'
Initializing USB-ISS in SPI mode with 146 spi-mode and 11 sck_divisor
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/py_opc-1.3.0-py.3.4.egg/opc/__init__.py", line 58, in __init__
IndexError: list index out of range

During handling of the above exception, another exception occured:

Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/py_opc-1.3.0-py.3.4.egg/opc/__init__.py", line 58, in __init__
IndexError: list index out of range

During handling of the above exception, another exception occured:

Please advise... is this a firmware issue?

dhhagan, 2017-06-30:
Hi @jeffisfast . That's interesting...do you know what firmware version your OPC is using? I'm not sure if Alphasense has released something since v18.2 or not..?

jeffisfast, 2017-06-30:
Hi @dhhagan Sorry, but I left out the line that says "The firmware version ix 0x7".  Does that help?

dhhagan, 2017-06-30:
Hmm. Not particularly. That means it is having a hard time reading the firmware version from the OPC. How have you hooked up the sensor? Via GPIO pins on the Pi? Or through a SPI-USB converter?

jeffisfast, 2017-06-30:
It is hooked up using the SPI-USB converter that came with the sensor.  Is there a way for me to talk directly to the USB port to query it?

dhhagan, 2017-06-30:
Okay. So you are using USB-ISS as outlined [here](http://py-opc.readthedocs.io/en/latest/#using-the-pyusbiss-library-via-usb-port)?

jeffisfast, 2017-06-30:
I just copied your code and here is the output when it runs, if this helps?

The module ID is 7
The firmware version is 0x7
The current operating mode is 0x92
The serial number is b'00034628'
Initializing USB-ISS in SPI mode with 146 spi_mode and 11 sck_divisor
/usr/local/lib/python3.4/dist-packages/py_opc-1.3.0-py3.4.egg/opc/__init__.py:524: UserWarning: Data transfer was incomplete.
Traceback (most recent call last):
  File "pm.py", line 15, in <module>
    for key, value in alpha.histogram().items():
AttributeError: 'NoneType' object has no attribute 'items'

For completeness here is the code I'm using:

import usbiss
import opc

spi = usbiss.USBISS("/dev/ttyACM0", 'spi', spi_mode = 2, freq = 500000)

try:
    alpha = opc.OPCN2(spi)
except Exception as e:
    print ("Startup Error: {}".format(e))

# Turn on the OPC
alpha.on()

# Read the histogram and print to console
for key, value in alpha.histogram().items():
    print ("Key: {}\tValue: {}".format(key, value))

# Shut down the opc
alpha.off()

jeffisfast, 2017-06-30:
Just as a follow up -- I switched power supplies to the sensor and now have gotten slightly further.  I've got a 5V DC power supply connected directly to the sensor, and then using their SPI-USB adapter to connect to the Raspberry Pi.  Here's the output from the code:

$ python3 pm.py
The module ID is 7
The firmware version is 0x7
The current operating mode is 0x92
The serial number is b'00034628'
Initializing USB-ISS in SPI mode with 146 spi_mode and 11 sck_divisor
Startup Error:
                        Your firmware version could not be automatically detected. This is usually caused
                        by a bad wiring or poor power supply. If niether of these are likely candidates, please
                        open an issue on the GitHub repository at https://github.com/dhhagan/py-opc/issues/new

Traceback (most recent call last):
  File "pm.py", line 12, in <module>
    alpha.on()
NameError: name 'alpha' is not defined

dhhagan, 2017-06-30:
Okay. So first things first. Are you sure that it is connecting to the correct serial port? It looks like you are using what was in the example code, which was what my Pi was connected to. It may or may not be the same for you. You can issue the command `ls /dev/tty*` to get a list of all connected devices.

jeffisfast, 2017-06-30:
I'm pretty sure it is on /dev/ttyACM0 as that device only appears while the sensor is plugged in. Any tips for how I can manually try communicating with the port to issue commands?

jeffisfast, 2017-06-30:
I tried using alpha = opc.OPC(spi) instead of OPC2 and am getting some communications:

>>> a.read_info_string()
'OPC-N2 FirmwareVer=OPC-018.2..............................BD'

It also seems that if I run things from the python3 interpreter they work, but running the script doesn't.  Will investigate further.

dhhagan, 2017-06-30:
Hmm. Maybe there is a permissions problem? Or try adding some time delays between statements?

jeffisfast, 2017-06-30:
It was a timing problem.  Throttling all the commands fixed everything.  Thanks for all the quick help!

closed 2017-06-30 by jeffisfast