dhhagan/py-opc #50: Create page for mentions/citations

dhhagan, 2017-12-02:
Allow people/orgs using the code to list themselves (if desired) so we can see who all is using this codebase!

DancingQuanta, 2017-12-03:
I have used this py-opc library for my summer placement at University of Leeds at Earth and Environment dept. I was studying the difference between customer-grade, industry-grade and scientific-grade aerosol sensors where an alphasense-N2 was evaluated.

dhhagan, 2017-12-03:
Hey @DancingQuanta, thanks for the update! I am considering submitting this repository to the Journal of Open Source Software, which means I would be listing you as a co-author since you are a contributor. Is that okay? If so, could you send me an email at dhagan@mit.edu with your most up-to-date affiliation/university/employer?

commit f15c429c (2018-01-19): close #50
closed 2018-01-19 by dhhagan