dhhagan/py-opc #55: How to connect to PM2.5 SPI version?

H60gh, 2018-01-19:
I connect a PM2.5 sensor into Reaspberry pi using SPI. Then I follow the example in Python 3.4.2, but it seems "alpha/alphasense" variable does not work and alpha.ping() returns "False". As I test, spidev works well.
I switch into Python 2.7.9 ("alpha = opc.OPCN2(spi)" is modified to "alpha = opc.OPC(spi))" and now alpha is an unknown variable.
How can I solve this problem?

dhhagan, 2018-01-19:
Hi there @H60gh,

If `alpha.ping()` is returning `False`, then the device is connected properly, but is not returning the correct value when the SPI command is sent. Have you tried any other commands? With Python2.7, try using the `OPCN2` class rather than the `OPC` class. Also, which version of the library are you using?

H60gh, 2018-01-19:
Hi @dhhagan 
I have same problem in Pytho 2.7 (Warning: Data Transfer was incomplete). Also I test other commands likes "read_info_string", only blank lines are printed!

My using libraries are:
py-opc  (1.4.1)
spidev (3.0)

H60gh, 2018-01-19:
In addition, I am using GPIO pins.

H60gh, 2018-01-19:
Ok. I add one minute sleep and it works. But how can I interpret return values and finally return pm1, pm2.5 and pm10 values?

dhhagan, 2018-01-19:
Hmm. Have you tried multiple readings? How are you wiring up the device? Can you post an image or wiring diagram?

H60gh, 2018-01-19:
Thank you Hagan. After multiple reading, I receive PM1, PM2.5 and PM10 values. May I ask another question? Temperature and Pressure values are still zero. Why?

closed 2018-01-19 by H60gh

dhhagan, 2018-01-19:
No problem. Depending on the unit you purchased, it may not physically be installed. I think Alphasense stopped adding them a couple of years ago unless you special order it (not sure why...). I add an external RHT sensor (SHT31 by Sensirion).