dhhagan/py-opc #56: Frimware version could not be automatically detected!!!

H60gh, 2018-02-11:
Hi everybody
A few weeks ago, I had run PM2.5 sensor on Raspberry Pi and it worked fine. Today I checked it again, but it did not work! I receive following error message: "Your frimware version could not be automatically detected. This is usually caused by bad wiring or a poor power supply. If niether of these are likely candidated open an issue on the github ...". I checked the power connections, everything is OK.

Before of this, I have update Rasbian OS. Also I attached a BME280 sensor to my board too. Can anybody help me?

H60gh, 2018-02-11:
I have put "alpha=opc.OPCN2(spi) in a while loop as follows (to remove the previously mentioned startup error in working by PM2.5 sensor):

while True:
    try:
         alpha = opc.OPCN2(spi)
         break
   except Exception as e:
        print ("Error:{}".format(e))

but it still does not work. As I checked, alpha.histogram() does not return related items.

DancingQuanta, 2018-02-11:
- Can you explain how do you connect your sensor to the Raspberry Pi?
- What kind of update OS procedure did you use?
- What version of the python are you using? Have the OS update touched python?

H60gh, 2018-02-11:
I connect my sensor using GPIO based on pin assignment in 
http://py-opc.readthedocs.io/en/latest/

Nothing changes during the time (the sensor worked well a few weeks ago). 

Today I have update the OS using simple update command in terminal.
and finally I use python 3.4 (same as before).

DancingQuanta, 2018-02-11:
Maybe the OS update modified GPIO interface somehow, what is the update command did you use?
Did you do usual unplug and replug, power off and on?

H60gh, 2018-02-11:
For BME280, I had to install Adafruit package I part of this was about installing GPIO frimware as follows:
```
sudo apt-get update
sudo apt-get install build-essential python-pip python-dev python-smbus git
git clone https://github.com/adafruit/Adafruit_Python_GPIO.git
cd Adafruit_Python_GPIO
sudo python setup.py install
```
Is it possible that these installations made changes in my PM2.5 driver?

DancingQuanta, 2018-02-11:
Maybe.
I need to know more about how you setup your Raspberry Pi hardware-wise. In other words, did you connect BME280 to GPIOs as the same time as alphasense?

H60gh, 2018-02-11:
Yes. I use a prototyping P1 extension board connected on top of my Raspberry pi board. My alphasense sensor uses "GND, 5.0v,MOSI,MISO,CLK and CS0" pins and BME280 uses "SDA,SCL,3.3v and GND" pins on the extension board. This boards only uses first 26 pins of standard GPIO pins of Raspberry pi 3.

H60gh, 2018-02-11:
Yes, at the same time, BME280 was attached to the extension board, but it did not work. Today I used BME280 using Adafruit and after that I saw that Alphasense did not work.

DancingQuanta, 2018-02-11:
I am sorry but I did not understand your first sentence, due to many commas there. What I wanted to know are the hardware changes between your alphasense working to not working. 
For example, did your alphasense work when connected directly to GPIOs without extension board?

H60gh, 2018-02-11:
The hardware did not change. BME280 was attached to the board, but I did not setting up it. Today I set it up and after that Alphasense did not work!

DancingQuanta, 2018-02-11:
Okay, your recent software changes may have messed things a bit up for you. 
@dhhagan should know more about GPIOs than I do, let wait for him to pop by.

H60gh, 2018-02-11:
It is interesting that I turned off the Raspberry pi and after a while I turned it on again. Then I test the alphasense in Python terminal, it worked well. After that, I added one second sleep time before reading alpasens.histgram() and now my code run perfectly!!!

It seems Raspberry Pi has some performance problems and it could not act well in all situations.

Thank you @DancingQuanta for your kindly support.

dhhagan, 2018-02-11:
Hey @H60gh have you tried setting the firmware version manually to see if that is the only call that has an issue? If that doesn't work, could you provide the entire code snippet you're using (for ex., what is `spi`?) as well as a simple wiring diagram? I have seen several people run into issues when there is not enough current to drive the OPC-N2, which would happen if using the 5V GPIO pin rather than a separate power supply.

dhhagan, 2018-02-11:
Ahh, yea I've seen people also experience that. Could you open a new issue that suggests adding that note to the documentation? It may help others in the future. I'm not sure of the root cause, but I assume it has something to do with the i2c timing/i.e. not letting the fall complete before the next rise.

H60gh, 2018-02-11:
Hi @dhhagan 
Thanks for the reply. Ok, I'll open a new issue.

closed 2018-02-11 by H60gh