dhhagan/py-opc #58: Update USB-ISS SPI mode

DancingQuanta, 2018-02-24:
Hi, I decided to change the schema for SPI modes in pyusbiss.
USB-ISS sets the SPI modes through a number in the following table

| USB-ISS command | SPI mode |
|----------------------|------------|
| 0                             | 0               |
| 1                             | 2               |
| 2                             | 1               |
| 3                             | 3               |

The numbers do not match up.
I made an assumption that iusers are familiar with SPI and knows its modes then they are more likely to select the SPI modes over USB-ISS mode numbers.

So in your application your SPI mode is 1 and for USB-ISS it is number command 2. With my change, you only need to enter 1 in USB-ISS example with OPC.
My library will set USB-ISS to use SPI mode 1 correctly.

I hope this will be more friendly.

DancingQuanta, 2018-03-04:
I think I should do a pull request anyway.

dhhagan, 2018-03-04:
Hey @DancingQuanta Please do send a PR! I will do my best to review/approve and test tomorrow. I've dropped off the map a little bit recently, but will get it done quick this week.

closed 2018-03-12 by DancingQuanta