dhhagan/py-opc #60: Trouble in establishing communication

Zanati123, 2018-03-08:
Dear David, 
I am a 3rd year university student doing a small project requiring to interface the alphasense sensor with RPi. I am having difficulties to establish communication and was hoping you can help me to solve the issue and/or guide me through the process.

A brief summary of the steps taken so far:
-	Installed Raspbian on RPI 
-	Downloaded zips py-opc and pyusbiss to /home/pi
-	Unzipped it and installed via terminal sudo install 
-	Then created a Py file with the script suggested in py-opc
-	Running the said script has highlighted script errors

I adjusted Dancing Quanta’s pyusbiss-master/usbiss/_init_.py  line 55/60 syntax positioning with the if statements, as that was the first error. The next error was on line 7 of the suggested script, alpha=opc.OPCN2(spi) shows the following error:

_>>>The module ID is 7
The firmware version is 0x7
The current operating mode is 0x2
The serial number is b'00032122'
Initializing USB-ISS in SPI mode with 1 spi_mode and 11 sck_divisor
Traceback (most recent call last):
  File "/home/pi/opc2/opc/__init__.py", line 55, in __init__
    self.firmware['version']    = int(re.findall("\d{3}", self.read_info_string())[-1])
  File "/home/pi/opc2/opc/__init__.py", line 197, in read_info_string
    self.cnxn.xfer([0x3F])
  File "/home/pi/opc2/usbiss/__init__.py", line 146, in xfer
    raise RuntimeError('USB-ISS: Transmission Error')
RuntimeError: USB-ISS: Transmission Error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/opc2/opc/__init__.py", line 59, in __init__
    self.firmware['version'] = int(re.findall("\d{1}", self.read_info_string())[-1])
  File "/home/pi/opc2/opc/__init__.py", line 197, in read_info_string
    self.cnxn.xfer([0x3F])
  File "/home/pi/opc2/usbiss/__init__.py", line 146, in xfer
    raise RuntimeError('USB-ISS: Transmission Error')
RuntimeError: USB-ISS: Transmission Error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/opc2/test5.py", line 7, in <module>
    alpha = opc.OPCN2(spi)
  File "/home/pi/opc2/opc/__init__.py", line 238, in __init__
    super(OPCN2, self).__init__(spi_connection, model='N2', **kwargs)
  File "/home/pi/opc2/opc/__init__.py", line 64, in __init__
    raise FirmwareVersionError(msg)
opc.exceptions.FirmwareVersionError: 
                            Your firmware version could not be automatically detected. This is usually caused by bad wiring or a poor power supply. If niether of these are likely candidates, please open an issue on the GitHub repository at https://github.com/dhhagan/py-opc/issues/new.”_

I am running on python 3.4.2 and have triple checked the connection from Rpi to alphasense opc-n2. Do you have any advice?

Thank you,
Zanati123

DancingQuanta, 2018-03-08:
Hi. I am the author of `pyusbiss`. I made a patch with `if` mistake. Please can you test it?
Also I will update the documentation in py-opc about new changes to pyusbiss.
Please change
```spi_mode = 2```
to 
```spi_mode = 1```

DancingQuanta, 2018-03-12:
Hello, I have released a new version of `pyusbiss` 0.2.0 which you can install via `pip install pyusbiss`. This version is a major revision and so I have updated the instructions in https://github.com/dhhagan/py-opc/pull/61.

DancingQuanta, 2018-04-12:
Have you made any progress?

Zanati123, 2018-04-16:
Hi DancingQuanta,

Yes with your new release of pyusbiss 0.2.0, communication was achieved to the OPC sensor. spi_mode = 1 and setting the firmware to 18.2 made it work.

```
from usbiss.spi import SPI
import opc

spi=SPI("/dev/ttyACM0")
spi.mode=1
spi.max_speed_hz = 500000

alpha = opc.OPCN2(spi, firmware=(18.2))

print( "Communication Status:  ", alpha.ping())
```

closed 2018-04-17 by Zanati123