dhhagan/py-opc #66: Data transfer was incomplete.

rach8cww, 2018-05-11:
I'm having an issue while reading the histogram from my device:

```
Trying to connect to instrument ttyACM0 ...
Connected to instrument! <usbiss.spi.SPI object at 0x76a49710>
Connected to Alphasense OPC-N2v18.2
Device status:
{'LaserDACVal': 1, 'FanDACVal': 1, 'FanON': 1, 'LaserON': 1}
Turning laser and fan on... Done.
Device status:
{'LaserDACVal': 1, 'FanDACVal': 1, 'FanON': 1, 'LaserON': 1}
----------------------------------------------------------------------
         Failed while retrieving results
Could not load histogram
----------------------------------------------------------------------

/home/pi/.local/lib/python3.5/site-packages/opc/__init__.py:522: UserWarning: Data transfer was incomplete.
  warnings.warn("Data transfer was incomplete.")
```

Any ideas on what the issue could be?

tavurth, 2018-05-11:
https://github.com/rach8cww/OPC_Work

DancingQuanta, 2018-05-11:
Does that happen every time you tried to get the histogram?

rach8cww, 2018-05-11:
Yes it does

dhhagan, 2018-05-11:
Could you let us know which version of the library you are using, as well as what version of Python, and what firmware version of the OPC-N2 you are using?

DancingQuanta, 2018-05-12:
> Hi dhhagan, I was using the pyusbiss library using a simple pip3 install
> with python 3 and the OPC-N2 firmware version I have is 18.2.

What version of pyusbiss are you using?

rach8cww, 2018-05-13:
Sorry for the delay, I'm using pyusbiss 0.2.2

DancingQuanta, 2018-05-13:
Can you try pyusbiss 0.2.0?

rach8cww, 2018-05-14:
```
pi@raspberrypi:~/project $ pip3 freeze | grep pyusbiss
pyusbiss==0.2.0
```

Still no luck, fails with the same error message:
```
/home/pi/.local/lib/python3.5/site-packages/opc/__init__.py:522: UserWarning: Data transfer was incomplete.
  warnings.warn("Data transfer was incomplete.")
```

rach8cww, 2018-06-21:
Solved the issue - the instrument needs to have time to warm up before you ask for the histogram. I've added a few short sleep pauses in as well and is makes it run smoother. This is the output now:

Trying to connect to instrument ttyACM0 ...
Connected to instrument! <usbiss.spi.SPI object at 0x76a23cf0>
Connected to Alphasense OPC-N2v18.2
Alphasense instrument processing request
-----------------------------------------------------------------------
Device status:
{'FanON': 1, 'LaserDACVal': 1, 'FanDACVal': 1, 'LaserON': 1}
Device status:
{'FanON': 4, 'LaserDACVal': 4, 'FanDACVal': 4, 'LaserON': 4}
----------------------------------------------------------------------
2018-06-21 14:29:17
Key: PM2.5      Value: 0.29654639959335327
Key: Bin 6      Value: 0.02371237245465762
Key: Pressure   Value: None
Key: Bin7 MToF  Value: 11.666666666666666
Key: Bin 12     Value: 0.0
Key: Bin 10     Value: 0.0
Key: Bin 1      Value: 0.02371237245465762
Key: Bin 0      Value: 0.08299330359130166
Key: PM1        Value: 0.08018143475055695
Key: Bin 3      Value: 0.01185618622732881
Key: Bin3 MToF  Value: 21.0
Key: Bin 8      Value: 0.03556855868198643
Key: Bin 4      Value: 0.01185618622732881
Key: Bin 11     Value: 0.0
Key: Checksum   Value: 25
Key: Temperature        Value: None
Key: PM10       Value: 2.543792486190796
Key: Bin 5      Value: 0.03556855868198643
Key: Bin1 MToF  Value: 12.666666666666666
Key: SFR        Value: 2.678318738937378
Key: Sampling Period    Value: 31.491455078125
Key: Bin 13     Value: 0.0
Key: Bin 9      Value: 0.02371237245465762
Key: Bin 14     Value: 0.0
Key: Bin 15     Value: 0.0
Key: Bin5 MToF  Value: 12.333333333333334
Key: Bin 2      Value: 0.03556855868198643
Key: Bin 7      Value: 0.01185618622732881
----------------------------------------------------------------------
2018-06-21 14:29:19
Alphasense OPC-N2v18.2 - Instrument finished getting data

dhhagan, 2018-06-22:
@rach8cww Great to hear! I'll be sure to add a note to the docs about waiting a bit between commands. I did speak with Alphasense and they mentioned this may be a requirement...

closed 2018-06-22 by dhhagan

DancingQuanta, 2018-06-22:
Does alphasense have some more details on timings for each i2c commands and
probably timing diagrams?

dhhagan, 2018-06-22:
@DancingQuanta They do not - it's quite unfortunate. I do have quite a few screenshots (and data) for most of the commands via a Logic Analyzer, but that's about it. In addition, some of the commands aren't designed to actually be functional, which they say will be fixed on the OPC-N3 (i.e. the laser_off command doesn't actually do anything).

DancingQuanta, 2018-06-22:
Ah well that is a shame. Looks like we have to figure out what works. What
will your strategy be regarding issues raised here?

tavurth, 2018-06-25:
I'd suggest creating a method, which simply checks every 200ms until the device is ready.

```python
alpha = opc.OPCN2(spi, debug=True).await()
```

dhhagan, 2018-06-26:
@tavurth That sounds like a fantastic idea, though I'm not sure exactly how to implement it with this device. Do you have any thoughts?

tavurth, 2018-06-26:
@dhhagan Sure, I made a quick PR

DancingQuanta, 2018-06-30:
I have an idea. Can `histogram` and others check the bytes as they come in. If the Alphasense is not ready then when you read in byte they comes as `\x01`. It would be useful to confirm that byte does not occur at first byte of correct response. So if you detect `\x01` then you could repeat the command byte and test again.

tavurth, 2018-06-30:
@DancingQuanta That would be preferable:

```python
        command = 0x30

        # Send the command byte
        self.cnxn.xfer([command])

        # Wait 10 ms
        sleep(10e-3)

        # read the histogram
        for i in range(62):
            r = self.cnxn.xfer([0x00])[0]
            resp.append(r)
```

Here we see that we send a command byte, and then expect to read 62*`uint16bit` of information.

1) Can we cancel that query by just sending another command byte? (Perhaps we have to read until the end of the 62 bits)
2) How can we make this work for something other than histograms?

I'd be wary about interfacing with the bits directly as we have a good interface to work off of here, built by @dhhagan. However if it's more stable, then we should certainly do so.

Perhaps you could comment on the above two questions if you have more information than I?

DancingQuanta, 2018-07-01:
Alas, I do not know the details of Alphasensw’s spi.

On other note, I noticed that this library already have a ping function, this is very short spi command and can be used to check if the device is ready.

PR #72 (2018-06-27): Added await function to alpha
Closes #66