dhhagan/py-opc #70: SPI-USB Adapter with OPC-N2 example does not work with any python version

tshu, 2018-06-22:
Related to #69, running the example script for the device using `python3` results in

```pi@raspberrypi:~/py-opc $ python3 reviewTest.py
Traceback (most recent call last):
  File "reviewTest.py", line 1, in <module>
    from usbiss.spi import SPI
ImportError: No module named 'usbiss'
```

Attempting to use `python` results in

```pi@raspberrypi:~/py-opc $ python reviewTest.py
ERROR:opc:Could not parse the fimrware version from ????????????????????????????????????????????????????????????
Traceback (most recent call last):
  File "/home/pi/py-opc/opc/__init__.py", line 77, in __init__
    self.firmware['version'] = int(re.findall("\d{3}", infostring)[-1])
IndexError: list index out of range
ERROR:opc:Could not parse the fimrware version from ?
Traceback (most recent call last):
  File "/home/pi/py-opc/opc/__init__.py", line 77, in __init__
    self.firmware['version'] = int(re.findall("\d{3}", infostring)[-1])
IndexError: list index out of range
WARNING:opc:Data transfer was incomplete
```

in addition to the device's fan continually running until unplugged.

DancingQuanta, 2018-06-23:
The first error showed you haven't installed pyusbiss.
Did you execute pip install pyuabiss?

DancingQuanta, 2018-06-23:
Ah. If you wanted python 3 to work then execute pip3 install pyusbiss

DancingQuanta, 2018-06-25:
Your second issue after executing `python2` appears to be a bad communication. 
Can you connect your sensor to the adaptor and wait for a minute before executing the `test-device.py` script? We have numerous Github issues regarding the connection problems and one of the solutions was to wait between physically connecting the sensor and executing any commands.

DancingQuanta, 2018-06-25:
Oh, your traceback does show that you successfully connected to your sensor.
The `py-opc` library attempts to communicate with your sensor a number of times before it succeeded. Your traceback shows two failed attempts and third attempt is successful.

Your real problem is the last line `WARNING:opc:Data transfer was incomplete`. This was due to `test-device.py` executing the histogram command at

dhhagan, 2018-06-25:
@tshu Can you let me know which version of `py-opc` is installed (`opc.__version__`)? I thought I fixed several of these in the newest release after speaking with Alphasense on some of the issues. Thanks!

tshu, 2018-06-25:
Great, so both python2 and python3 work with test-device.py after using pip3 and waiting after USB connection. I will occasionally fail to connect 2-3 times, but it will always run the script in the end. If there can be a note about the wait time in the readme, that would be great.

However, referring back to the original problem in this issue of running the example code from the readme, both python2 and python3 will attempt to connect 0-3 times before terminating with `WARNING:opc:Data transfer was incomplete'. Fan remains on. I can still run test-device.py to regain control of the device and have the fans turn off at the end of the script.

@dhhagan currently running 1.6.0

DancingQuanta, 2018-06-25:
What is output of the following script (modified from readme)
```
import spidev
import opc
from time import sleep

spi = spidev.SpiDev()
spi.open(0, 0)
spi.mode = 1
spi.max_speed_hz = 500000

alphasense = opc.OPCN2(spi)

# Turn the opc ON
alphasense.on()

# Read the information string
print (alphasense.read_info_string())

# Turn the opc OFF
alphasense.off()
```

tshu, 2018-06-25:
```
pi@raspberrypi:~/py-opc $ python3 reviewTest2.py 
Traceback (most recent call last):
  File "reviewTest2.py", line 14, in <module>
    alphasense.on()
NameError: name 'alphasense' is not defined
pi@raspberrypi:~/py-opc $ python reviewTest2.py 
Traceback (most recent call last):
  File "reviewTest2.py", line 14, in <module>
    alphasense.on()
NameError: name 'alphasense' is not defined
```

dhhagan, 2018-06-25:
@DancingQuanta @tshu It would appear the script/code above is incorrect - it should be `alpha.on()` since that is what the `OPCN2` class instance was initiated as. I have just merged the previous readme changes into `v1.6.1`. Care to check?

DancingQuanta, 2018-06-25:
Opps, I made a name mistake.

```
import usbiss
from usbiss.spi import SPI
import opc
from time import sleep

spi = SPI("/dev/ttyACM0")

spi.mode = 1
spi.max_speed_hz = 500000

alpha = opc.OPCN2(spi, debug=True)

# Turn the opc ON
alpha.on()

# Read the information string
print (alpha.read_info_string())

# Turn the opc OFF
alpha.off()
```

tshu, 2018-06-25:
```
pi@raspberrypi:~/py-opc $ python reviewTest2.py 
ERROR:opc:Could not parse the fimrware version from ?
Traceback (most recent call last):
  File "/home/pi/py-opc/opc/__init__.py", line 77, in __init__
    self.firmware['version'] = int(re.findall("\d{3}", infostring)[-1])
IndexError: list index out of range

pi@raspberrypi:~/py-opc $ python3 reviewTest2.py 
ERROR:opc:Could not parse the fimrware version from ?
Traceback (most recent call last):
  File "/home/pi/py-opc/opc/__init__.py", line 77, in __init__
    self.firmware['version'] = int(re.findall("\d{3}", infostring)[-1])
IndexError: list index out of range

```

Getting some garbage outputs.

DancingQuanta, 2018-06-25:
Could you replace `print(alpha.read_info_string())` with `print(repr(alpha.read_info_string()))` and try again  please?

tshu, 2018-06-25:
```
pi@raspberrypi:~/py-opc $ python reviewTest2.py 
ERROR:opc:Could not parse the fimrware version from ?
Traceback (most recent call last):
  File "/home/pi/py-opc/opc/__init__.py", line 77, in __init__
    self.firmware['version'] = int(re.findall("\d{3}", infostring)[-1])
IndexError: list index out of range
'\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01'
pi@raspberrypi:~/py-opc $ python3 reviewTest2.py 
ERROR:opc:Could not parse the fimrware version from ?
Traceback (most recent call last):
  File "/home/pi/py-opc/opc/__init__.py", line 77, in __init__
    self.firmware['version'] = int(re.findall("\d{3}", infostring)[-1])
IndexError: list index out of range
'\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01'
```

DancingQuanta, 2018-06-25:
Your output shows that it is getting blank data from your sensor.
Can you add `sleep(1)` before each line containing `alpha` and run the script?

tshu, 2018-06-25:
That did the trick.

```
pi@raspberrypi:~/py-opc $ python3 reviewTest2.py 
'OPC-N2 FirmwareVer=OPC-018.2..............................BD'
pi@raspberrypi:~/py-opc $ python reviewTest2.py 
'OPC-N2 FirmwareVer=OPC-018.2..............................BD'
```
The only necessary sleep is between turnon and the read. The following code works:

```
import usbiss
from usbiss.spi import SPI
import opc
from time import sleep

spi = SPI("/dev/ttyACM0")

spi.mode = 1
spi.max_speed_hz = 500000

alpha = opc.OPCN2(spi, debug=True)

# Turn the opc ON
alpha.on()

sleep(1)
# Read the information string
print(repr(alpha.read_info_string()))

# Turn the opc OFF
alpha.off()
```

DancingQuanta, 2018-06-25:
We probably need to improve `alpha.on()`. A sleep could be added to end of `alpha.on()`. Or have a timer in the library that track which command have been sent. So that the script itself can work without being slowed down until it execute another `alpha` command which then will invoke sleep with time period equal to alphasense's response then execute. Like a debt.

dhhagan, 2018-06-26:
@DancingQuanta Yea, I tried to avoid doing anything other than wrapping the SPI commands, but it seems that it is causing so much trouble for most users (myself included). I went ahead and added sleeps in the necessary places and will push a bit later on today as I have some free time.

DancingQuanta, 2018-06-27:
Can I suggest a decorator to wrap some methods to add sleep to beginning of a method if the method was called in the specified time period since last method called. Like a cool down. We know that a method such as `on` takes a while to make the device ready.
Every time a method such as `on` is called the time of this call is recorded. Next time a method (any method) is called, the new time is compared to old time and the difference is used to delay execution of called method.

This way the time spent waiting for `sleep` in each method to work would be less if we could use the time in between calls as part of waiting.

dhhagan, 2018-06-27:
@DancingQuanta I'll go ahead and merge a few PR's and push right now - maybe then take a look and see what you think is best? A decorator would be fine as you suggest - I think I just added a few sleeps. Give me 30 min or so.

dhhagan, 2018-06-27:
@tshu I've added some notes to the readme. Should be okay now?

closed 2018-06-29 by dhhagan