dhhagan/py-opc #75: Documentation clarification

selimnairb, 2018-07-11:
For `opc.OPCN2.pm()` and `opc.OPCN2.histogram()` functions, PM1, PM2.5 and PM10 values are reported.  I am assuming the units on these are μg/m3, however the documentation doesn't say.  The documentation should be updated to clearly state the units that the data are being reported in.

dhhagan, 2018-07-27:
Thanks for the suggestion @selimnairb. I will add this into the next release.

yash-dahima, 2023-09-20:
Hi @dhhagan,

Thanks for developing this wonderful library. I need to clarify something from the documentation. It will be of a great help if you could help me with this.

I was reading documentation on "https://py-opc.readthedocs.io/en/latest/". The section "Connecting via GPIO" mentions "The OPC requires at least 250 mA, so powering directly through the RPi is not an option". Please explain me this as I do not have much knowledge of electronics.

Can't RPi provide this current? If not, please give me an idea of how to connect both RPi and OPC-N3 with the same power supply. Can both the device be powered using mobile charging adapter?

Thanks for your time! Waiting for your response...