dhhagan/py-opc #76: ERROR: ]

mc13756, 2018-07-16:


dhhagan, 2018-07-16:
Hi @mc13756 The error comes from not being able to correctly parse the firmware version of the OPC. Often, this is caused by a wiring issue, power issue, or sometimes a code issue. Can you please let us know which version of the software you are using (`opc.__version__`) and post the code you used to get to this point?

dhhagan, 2018-07-16:
Thanks for that information. Which version of the `py-opc` library are you using? You can get it by issuing the command `opc.__version__`; you can also get it by running `$pip list` at the terminal and finding the version number of the package.

mc13756, 2018-07-16:
I am using the library py-opc (1.6.0)

dhhagan, 2018-07-16:
Thanks. How is it wired up? I assume you are using the GPIO pins? How is it being powered?

mc13756, 2018-07-16:
Yes, I am using the GPIO pins. I give 4.6 V through an external battery. I've already checked the wiring connection and it's working fine.

dhhagan, 2018-07-16:
Ahh okay. Have you verified the OPC-N2 will work with 4.6V? I believe it requires at least 4.8VDC and > 250 mA. You could test it by simply plugging the battery into the micro-USB slot on the OPC; it will start up by itself after a minute and log data internally. You can then grab that data via the micro-USB slot and see if it makes sense.

mc13756, 2018-07-16:
I increased the voltage to 4.9 V with the same result. Also, I tried the following code too, in order to get the results analytically:

https://gist.github.com/dhhagan/bfb17459f0067a120f09?signup=true

Sometimes the code starts compiling and I get an error at line 83 and sometimes I have the same result as the previous code. 
Error: opc: Firmware version is invalid for this device.

dhhagan, 2018-07-16:
Hmm. Can you draw the wiring diagram. It's fairly easy to mix up MISO and MOSI, so I'd like to rule that out..

mc13756, 2018-07-17:
[image]

metallicano, 2018-07-19:
how can i now what firmaware version am using?

dhhagan, 2018-07-20:
Hi @metallicano; are you able to connect to the device? If so, you can use either `read_information_string()` or `read_firmware_version()`.

dhhagan, 2018-07-20:
@mc13756 This seems like a wiring issue (hopefully an easy fix!). According to this [diagram][1], you should have the following:

  * SCK -> GPIO 14, Pin 23
  * SDO/MISO -> GPIO 13, Pin 21
  * SDI/MOSI -> GPIO 12, Pin 19
  * /SS -> GPIO 10 or 11, Pins 24, 26

If you haven't already seen this, the instructions can be found [here][2].

Let me know if this doesn't work.

[1]: http://pi4j.com/pins/model-3b-rev1.html
[2]: http://py-opc.readthedocs.io/en/latest/#connecting-via-gpio

mc13756, 2018-07-20:
I've checked the wiring and is just like you mentioned above.  I run the extended version of code
https://gist.github.com/dhhagan/bfb17459f0067a120f09?signup=true
and I got the following error. 

ERROR:opc:Firmware version is invalid for this device.
Traceback (most recent call last):
  File "/home/pi/Desktop/opc_test_3.py", line 31, in <module>
    alpha = OPCN2(spi)
  File "/home/pi/.local/lib/python3.5/site-packages/opc/__init__.py", line 301, in __init__
    raise FirmwareVersionError("Your firmware is not yet supported. Only versions 14-18 are currently supported.")
opc.exceptions.FirmwareVersionError: Your firmware is not yet supported. Only versions 14-18 are currently supported

Also, when I tried to install py-opc by typing:    pip install py-opc [--upgrade]   in a terminal I got:
"Invalid requirement: '[--upgrade]'
so I've installed it by typing only: pip install py-opc

dhhagan, 2018-07-20:
@mc13756 Try `pip install py-opc --upgrade` - the brackets are just to indicate it is optional

mc13756, 2018-07-20:
ok. thank you.
It seems like OPC works randomly.  Sometimes I got the following error:

ERROR:opc:Firmware version is invalid for this device.
Traceback (most recent call last):
  File "/home/pi/Desktop/opc_test_3.py", line 31, in <module>
    alpha = OPCN2(spi)
  File "/home/pi/.local/lib/python3.5/site-packages/opc/__init__.py", line 301, in __init__
    raise FirmwareVersionError("Your firmware is not yet supported. Only versions 14-18 are currently supported.")
opc.exceptions.FirmwareVersionError: Your firmware is not yet supported. Only versions 14-18 are currently supported

Sometimes the code runs normally and I got numbers every 10 seconds, and some other times it looks like it runs ok but I got only zeros. In this case the OPC's fan isn't working and I get false on turning ON or OFF the opc.

Turning on the opc
ON: False

or

Turning off the opc
OFF: False

dhhagan, 2018-07-20:
@mc13756 Interesting - the script your using is pretty outdated (from 2015, I think). I would take a look at the most up-to-date docs for reference. When did you buy the OPC and do you have the SPI-USB connector for debugging purposes? Your OPC is most likely v18.2, but I can't be sure..

mc13756, 2018-07-23:
I think we bought the OPC on 2015 and our version is 18.2. Yes, I have the SPI-USB connentor

metallicano, 2018-07-23:
hi , the problem persist , but in the end of the text i have this : your firmware is not supported :C

metallicano, 2018-07-23:
now the code runs but sometimes XD,

dhhagan, 2018-07-23:
@metallicano Can you post the code and wiring you're using?

metallicano, 2018-07-23:
import spidev
import opc
from time import sleep

spi = spidev.SpiDev()
spi.open(0, 0)
spi.mode = 1
spi.max_speed_hz = 500000

alphasense = opc.OPCN2(spi)

# Turn the opc ON
alphasense.on()

sleep(1)

# Read the information string
print (alphasense.read_info_string())

# Read the histogram
print (alphasense.histogram())

# Turn the opc OFF
alphasense.off()

thw wiring is the same : 
SCK -> GPIO 14, Pin 23
SDO/MISO -> GPIO 13, Pin 21
SDI/MOSI -> GPIO 12, Pin 19
/SS -> GPIO 10 or 11, Pins 24, 26

metallicano, 2018-07-23:
import spidev
import opc
from time import sleep

spi = spidev.SpiDev()
spi.open(0, 0)
spi.mode = 1
spi.max_speed_hz = 500000

alphasense = opc.OPCN2(spi)

# Turn the opc ON
alphasense.on()

sleep(1)

# Read the information string
print (alphasense.read_info_string())

# Read the histogram
print (alphasense.histogram())

# Turn the opc OFF
alphasense.off()

metallicano, 2018-07-23:
import spidev
import opc
from time import sleep

spi = spidev.SpiDev()
spi.open(0, 0)
spi.mode = 1
spi.max_speed_hz = 500000

alphasense = opc.OPCN2(spi)
alphasense.on()
sleep(1)
print (alphasense.read_info_string())
print (alphasense.histogram())
alphasense.off()

metallicano, 2018-07-23:
the wiring is the same : 
SCK -> GPIO 14, Pin 23
SDO/MISO -> GPIO 13, Pin 21
SDI/MOSI -> GPIO 12, Pin 19
/SS -> GPIO 10 or 11, Pins 24, 26

metallicano, 2018-07-23:
hi , problem solved , thanks for all 💯

DancingQuanta, 2018-07-26:
The decimals could be an average. Can you show the data you are getting please?

dhhagan, 2018-07-26:
Hey @mc13756 The reason you are getting decimals is because the default argument for `number_concentration` is `True` (read more about it in the [docs](http://py-opc.readthedocs.io/en/latest/#opc.OPCN2.histogram)). Thus, the number concentration in units of #/cc is being returned rather than the raw count. If you want the raw count, simply set `number_concentration=False`.

mc13756, 2018-07-26:
How can I get access to change the number_concentration to False?

dhhagan, 2018-07-26:
@mc13756 It's just an argument to `opc.histogram()`. So if you wanted to return the values as a raw histogram, you would do `opc.histogram(False)` or `opc.histogram(number_concentration=False)`.

mc13756, 2018-07-27:
I changed the argument to False and everything looks to work normally. 

Thank you very much!!!

dhhagan, 2018-07-27:
Good to hear, @mc13756: just be aware that the number you are seeing is the total number of particles in each bin since the last time you called either `pm()` or `histogram()`. In doing the conversion to #/cc, I was accounting for the sampling time and flow rate.

closed 2018-07-27 by dhhagan