dhhagan/py-opc #78: Add support for the OPC-N3

dhhagan, 2018-07-26:
I have just received an OPC-N3 and can thus begin this process!

Mi-Ch, 2018-10-16:
Have started work on support for OPC-N3 in py-opc?
//Thank you for your great project!

dhhagan, 2018-10-16:
Hi @Mi-Ch I have not yet, but I can if there is a demand for it...I'll try and get started within the next week or so.

Mi-Ch, 2018-10-16:
Thank you @dhhagan ! That would be great. We just received two OPC-N3 and we are setting up small comparison study with PMS7003 (x2) and SEN0177 (x2).

biogis, 2018-10-17:
Hello,
We'll also need OPC-N3 support soon. We worked with the N2 but decide to go further with the last sensor.
Great project also, thanks for the job you are doing.

dhhagan, 2018-10-17:
Oye - I'll do my best...have a bunch of papers to write and a dissertation to finish up, so hold on...ha

Mi-Ch, 2018-10-22:
@dhhagan I know the pain, I'm in the same situation ;(. Dissertations, papers....but preparing new measurements/experiments is much more interesting than writing about old ones.

dhhagan, 2018-10-22:
Alright @Mi-Ch and @biogis - I will get to this this week (hopefully sometime tomorrow). In the meantime, it would be great if you (and anyone else reading this) took a second to add to the [mentions page][1] to let me/others know how you're using this software. If nothing else, maybe it will convince some other PI's that releasing OSS isn't a huge waste of time!

[1]: https://github.com/dhhagan/py-opc/blob/master/MENTIONS.md

dhhagan, 2018-11-01:
Alright - an update: I just set aside some time to implement this into the package, but the Alphasense-provided documentation on the SPI protocol is totally incomprehensible. I would suggest anyone that needs this to operate the Alphasense OPC-N3 reach out to Alphasense directly and ask them to either add it themselves, clarify their documentation, or hire someone to do it. I apologize that I wasn't able to get this working for you all, but I just don't have the bandwidth right now to deal with them.

biogis, 2018-11-02:
It might be a dummy question, but the failing connection is due to the spi or to another firmware in the opc which is not recognised py the opc.OPCN2 function?

My first error is by opening the connection using the opc.OPCN2() function, and it mentions a firmware error.

dhhagan, 2018-11-02:
@biogis Yea, I would say that's expected. When you initiate an instance of OPCN2, it automatically issues the `read_info_string()` method to get the firmware version so that it knows which other methods are unavailable. For the OPCN2's, the firmware versions were in the range 15-18.2; however, on the OPCN3's, they switched it up, and now the firmware versions are stuck in v1 (i.e. v1.17) - not sure why they did that...

Anyways, as far as I can tell, they changed all of the SPI commands and added an extra command byte which means that none of the old code will work - some of it is similar, but the one I don't understand is how to turn the device on and off, which is kind of important...

Velivole, 2018-11-07:
Hi everybody, 
I just wanted tu port my project from the N2 to the N3 an stumbled upon some errors and then this thread. This doesn't sound very good, what Alphasense did there. 

I am not that kind of pro to work on this github project actively. But is there anything I can do (i.e. contact Alphasense) to help make things work? We are really dependened on a working N3.

Thank you very much!

biogis, 2018-11-08:
Simple and single answer I could get from Alphasense is:
"The PSI commands etc for the OPC are similar but not identical to the N2"

And I got the documentation. I'll see what I can do and change on my side

Velivole, 2019-01-25:
Hi everybody,
my PhD supervisor managed to get a bit of th py-opc code working with the Alphasense OPC-N3. However, there is still some work to do and we are hopeing for any help, that can be contributed. You can find the fork here: https://github.com/philipan/py-opc

dhhagan, 2019-01-25:
@Velivole This is great! Is there anything specific you think needs to be looked at? I just looked very briefly and it looks like it was written for python2 - prior to merging with this repo it should support python3, but that's an easy change. 

I'm a bit short no time right now, but can look over specific things if you would like.

PlanningChris, 2019-05-29:
I'm just starting out on a project to connect the OPC-N3 to a Pi and came across this. Did you ever get any further working with the OPC-N3? It looks like the fork mentioned by Velivole is the way to go for now?

dhhagan, 2019-05-29:
Hey @PlanningChris, I have not had time to take a look at it yet, so yes, that is probably the way to go for now. If anyone gets a version working and confirmed, I am more than happy to take a look at pull requests. I will update this thread if I can find the time to take a look and build a working version!

lizcorson, 2020-10-12:
@Velivole @dhhagan Are you still interested in contributions? I have successfully written a function to calculate the checksum for the OPC-N3 based on the fork @Velivole linked. My repository is here for your review before I submit a pull request: https://github.com/lizcorson/py-opc/tree/opc-n3

dhhagan, 2020-10-12:
Hi @lizcorson yes, contributions are still welcome. I've been quite limited in time, but do have some N3's lying around that I can use to test now...

lizcorson, 2020-10-12:
Great! I also changed a few things over to Python 3 as well. I had to do that for my own testing. device-test.py uses the OPCN3 class now.

phoebebright, 2020-10-19:
Just used @lizcorson branch and working for me also.  Thanks Liz :-)

dhhagan, 2020-10-19:
Great to hear @phoebebright. @lizcorson If you want to submit a PR, I can merge it in. I haven't looked through everything yet, but generally speaking, we will just want to make sure the code passes unit tests and has any necessary documentation. i can certainly help with that or provide guidance if needed. If these steps are already done, then feel free to ignore and just submit the PR.

lizcorson, 2020-10-19:
I have not yet run the unit tests so I'll give that a try and let you know when it's ready. Thanks, @dhhagan!

fargiolas, 2022-05-19:
Hey, if anyone is interested I'm open sourcing a library that supports all new OPC devices (N3, R1 and R2). It was written from scratch trying to abstract the common stuff in the different OPC versions into a single interface. Contribution, testing, issues and PRs are more than welcome!

You can find it at [https://github.com/fargiolas/py-opc-ng](https://github.com/fargiolas/py-opc-ng)