dhhagan/py-opc #81: ERROR:opc:Could not parse the fimrware version from  Traceback (most recent call last):

tomasfichetti, 2018-12-03:
Hello! I'm having problems with opc-n2. I already check the cables and the power. I also update the library.

This is my program:

 import spidev
import opc
from time import sleep

spi = spidev.SpiDev()
spi.open(0, 0)
spi.mode = 1
spi.max_speed_hz = 500000

alphasense = opc.OPCN2(spi)

# Turn the opc ON
alphasense.on()

sleep(1)

# Read the information string
print (alphasense.read_info_string())

# Read the histogram
print (alphasense.histogram())

# Turn the opc OFF
alphasense.off()

**The return is:**

ERROR:opc:Could not parse the fimrware version from 
Traceback (most recent call last):
  File "/home/pi/.local/lib/python2.7/site-packages/opc/__init__.py", line 77, in __init__
    self.firmware['version'] = int(re.findall("\d{3}", infostring)[-1])
IndexError: list index out of range
ERROR:opc:Could not parse the fimrware version from 
Traceback (most recent call last):
  File "/home/pi/.local/lib/python2.7/site-packages/opc/__init__.py", line 77, in __init__
    self.firmware['version'] = int(re.findall("\d{3}", infostring)[-1])
IndexError: list index out of range
ERROR:opc:Could not parse the fimrware version from 
Traceback (most recent call last):
  File "/home/pi/.local/lib/python2.7/site-packages/opc/__init__.py", line 77, in __init__
    self.firmware['version'] = int(re.findall("\d{3}", infostring)[-1])
IndexError: list index out of range
ERROR:opc:Could not parse the fimrware version from 
Traceback (most recent call last):
  File "/home/pi/.local/lib/python2.7/site-packages/opc/__init__.py", line 77, in __init__
    self.firmware['version'] = int(re.findall("\d{3}", infostring)[-1])
IndexError: list index out of range
ERROR:opc:Could not parse the fimrware version from 
Traceback (most recent call last):
  File "/home/pi/.local/lib/python2.7/site-packages/opc/__init__.py", line 77, in __init__
    self.firmware['version'] = int(re.findall("\d{3}", infostring)[-1])
IndexError: list index out of range
ERROR:opc:Could not parse the fimrware version from 
Traceback (most recent call last):
  File "/home/pi/.local/lib/python2.7/site-packages/opc/__init__.py", line 77, in __init__
    self.firmware['version'] = int(re.findall("\d{3}", infostring)[-1])
IndexError: list index out of range
Traceback (most recent call last):
  File "opc_spi_basico.py", line 10, in <module>
    alphasense = opc.OPCN2(spi)
  File "/home/pi/.local/lib/python2.7/site-packages/opc/__init__.py", line 293, in __init__
    super(OPCN2, self).__init__(spi_connection, model='N2', **kwargs)
  File "/home/pi/.local/lib/python2.7/site-packages/opc/__init__.py", line 71, in __init__
    raise FirmwareVersionError(msg)
opc.exceptions.FirmwareVersionError: 
                            Your firmware version could not be automatically detected. This is usually caused by bad wiring or a poor power supply. If niether of these are likely candidates, please open an issue on the GitHub repository at https://github.com/dhhagan/py-opc/issues/new. Another option would be to
                            increase the max_cnxn_retries variable if you feel the serial communication is being held up for some reason.

dhhagan, 2020-06-12:
Hi @tomasfichetti Is this still an issue? Sorry for the long delay here...

dwitam, 2020-12-19:
Hello, Getting the same issue. Can you help me out.. I am using OPC-R1.

ERROR:opc:Could not parse the fimrware version from 1ó1ó1ó1ó1ó1ó1ó1ó1ó1ó1ó1ó1ó1ó1ó1ó1ó1ó1ó1ó1ó1ó1ó1ó1ó1ó1ó1ó1ó1ó
Traceback (most recent call last):
  File "/home/pi/py-opcr1/opc/__init__.py", line 77, in __init__
    self.firmware['version'] = int(re.findall("\d{3}", infostring)[-1])
IndexError: list index out of range

                            Your firmware version could not be automatically detected. This is usually caused by bad wiring or a poor power supply. If niether of these are likely candidates, please open an issue on the GitHub repository at https://github.com/dhhagan/py-opc/issues/new. Another option would be to
                            increase the max_cnxn_retries variable if you feel the serial communication is being held up for some reason.

dhhagan, 2020-12-19:
Hi @dwitam can you post the code you're using? There currently isn't support for the OPC-R1 or OPC-N3 in the master branch as they're completely different. There is a development branch that supports (hopefully) these sensors, but it hasn't been tested.

dwitam, 2020-12-21:
`import spidev
import opc
import time

#? Open a  SPI connection on CE0
spi = spidev.SpiDev()
spi.open(0,0)

#? set the SPI Mode and clock speed
spi.mode=1
spi.max_speed_hz= 500000

try:
        alpha=opc.OPCR1(spi)

except Exception as e:
    print(e)

alpha.on()
time.sleep(1)

print("<---------Reading Firmware version-------------->")
print(alpha.read_firmware())

print("Read info string")
print(alpha.read_info_string())

print("Read serial number")
print(alpha.sn())

print("Read Histogram")
print(alpha.histogram())

time.sleep(10)

print("Read PM")
print(alpha.pm())
print("Pinging From RPI")
print(alpha.ping())

#time.sleep(10)
# Turn the device off
alpha.off()
`