dhhagan/py-opc #83: why i can`t supply the OPC-N2 with the raspberry ?

metallicano, 2019-03-16:


dhhagan, 2019-03-18:
Hi @metallicano Can you be more specific so I can help? Thanks!

metallicano, 2019-03-19:
in the documentation of the py-opc`s is it : The OPC requires at least 250 mA, so powering directly through the RPi is not an option....why?

dhhagan, 2019-03-19:
Ahh okay. The new RPi's can only output 100 mA from all GPIO pins combined, whereas the older models can only output 50 mA total (16mA per pin max).

You can try it from the 5V pin, but according to the specs, it shouldn't work..

patricktokeeffe, 2019-03-21:
@dhhagan I think you may have misread the [specs page](https://www.raspberrypi.org/documentation/hardware/raspberrypi/power/README.md). I/O pins are limited to 50mA but the 5V pin is tied directly to the power rail. It can source whatever the power supply can provide, certainly over 1A. See also:

* https://raspberrypi.stackexchange.com/questions/9298/what-is-the-maximum-current-the-gpio-pins-can-output
* https://raspberrypi.stackexchange.com/questions/1617/how-do-i-supply-power-through-the-gpio
* https://pinout.xyz/pinout/pin2_5v_power

Standard procedure in our university research lab is to power the OPC from 5V & G pins to avoid potential grounding offset issues that cause SPI comms to choke. I've had units running for multiple years in this configuration.

dhhagan, 2019-03-21:
@patricktokeeffe ahh okay - good to know! I'll change the docs then when I have time unless someone wants to send a PR with the changes..

patricktokeeffe, 2019-03-21:
@dhhagan I'd be happy to submit a PR later tonight

metallicano, 2019-03-21:
SO, i can connect the opc to 5V raspberry  pin and it will work ok ?

patricktokeeffe, 2019-03-29:
Hi @metallicano you tried it out? if so how is it working?

metallicano, 2019-05-13:
Yes, it works

closed 2019-05-13 by dhhagan

metallicano, 2019-05-13:
yes it is working by the moment "good", i will to compare it whith  other
sensor, the DUSTTRAK 8533 and then i'll tell you the results.