dhhagan/py-opc #85: Turn off the Fan of OPC-R1

cobalt67, 2020-06-12:
I am running OPC-R1 using SPI to USB and i have connected it to Raspberry Pi.
When I use alpha.off(), it stops the sensing but does not turn off the Fan of the sensor. The Fan keeps on going, even when the script is not running.

Could you suggest how should I turn off the Fan of OPC-R1?

dhhagan, 2020-06-12:
Hi @cobalt67 This library currently does not support the OPC-N3 or OPC-R1 - just the OPC-N2. Many have tried to adapt it for the newer sensors with mixed reviews, and unfortunately, I don't have those devices to test, so I haven't been able to work on it myself. I believe if you search through the issues you will find some on the topic of the R1 and N3 that may be helpful.

cobalt67, 2020-06-18:
Hey @dhhagan Thanks a lot for the information. My R1 works well with Rpi, but the results are not that satisfactory. Probably because R1 is not that reliable. Could you please suggest some other lightweight PM sensor, I basically integrate it on a UAV?

Also, could you please explain why does b1 == 0xF3 and b2 == 0x03 in class OPCN2(_OPC) ?

>>> alpha.off()
        True
        """
        b1 = self.cnxn.xfer([0x03])[0]          # send the command byte
        sleep(9e-3)                             # sleep for 9 ms
        b2 = self.cnxn.xfer([0x01])[0]          # send the following two bytes
        sleep(0.1)

        return True if b1 == 0xF3 and b2 == 0x03 else False

FlorentinBulotUoS, 2020-06-18:
Hi @cobalt67 

This comes from the manual of the OPC-N2. After sending 0x03 it should return 0xF3 and after sending 0x01, it should return 0x03.

The OPC-R1 has different command bytes to the OPC-R1 so the class OPCN2(_OPC) will not work with it. You can try my fork of the code for the implementation of the OPCR1 class: https://github.com/FlorentinBulotUoS/py-opc I have been using it for a year or so on a number of OPCR1 and it is working fine but I am looking for someone else to test it. 

I hope that helps,
Best wishes.

cobalt67, 2020-06-18:
Hey @FlorentinBulotUoS 
Thanks for clearing it out. I tried your OPCR1 class, and my issue of fan and laser is solved now! Thanks a lot for sharing it, your py-opc works perfectly with my R1.

One more thing, I have been using OPCN2(_OPC) class till now with R1 and it is also working with R1, I am not sure how it is compatible with the bytes of R1?