dhhagan/py-opc #86: Help: basic operation to read from OPC N2

abhatikar, 2021-05-13:
pyusbiss version: 0.2.2
Python version: 3.6.9
Operating System: Ubuntu 18.04
Description
Describe what you were trying to get done.
I am trying to read readings from OPC N2 sensor via SPI
I am using the library https://github.com/dhhagan/py-opc which using this library to access the OPC N2.

Tell us what happened, what went wrong, and what you expected to happen.
I am getting this error when I run the test

python test-device.py
Traceback (most recent call last):
File "test-device.py", line 10, in
spi = SPI("/dev/ttyUSB0")
File "/root/spi_dev/lib/python3.6/site-packages/usbiss/spi.py", line 16, in init
self._usbiss = USBISS(port)
File "/root/spi_dev/lib/python3.6/site-packages/usbiss/usbiss.py", line 63, in init
self.get_iss_info()
File "/root/spi_dev/lib/python3.6/site-packages/usbiss/usbiss.py", line 110, in get_iss_info
raise USBISSError("Could not get version details")
usbiss.usbiss.USBISSError: Could not get version details

What I Did
I have connected OPC n2 via spi to adafruit ft232H board.
Connections
OPC FT232H

5V <------> 5V
GND <------> GND
MOSI <------> D1
MISO <------> D2
SCK <------> D0
SS <------> C0

Any help would be appreciated..!

dhhagan, 2021-05-13:
Hi @abhatikar - can you include the code you're using to setup the SPI connection and OPC instance?

abhatikar, 2021-05-13:
Thanks for the reply. I am using the test-device.py from the repo. @dhhagan

dhhagan, 2021-05-14:
@abhatikar Okay - the error indicates there is something wrong with the wiring. In this case, it is likely caused by the intermediary board - I'm not familiar with the FT232H board and don't really have the time to dig into it today, but do you know if there is anything special you need to do for setup? Are you using the correct port (`/dev/tty/USB0`)?

abhatikar, 2021-05-14:
Hello @dhhagan , I really appreciate your work and efforts in maintaining this library for makers. I also appreciate you are busy, but if we get on a quick call at your said time, I could go over the full setup. Let me know. Thanks in advance.

abhatikar, 2021-05-14:
> @abhatikar Okay - the error indicates there is something wrong with the wiring. In this case, it is likely caused by the intermediary board - I'm not familiar with the FT232H board and don't really [...]

Yes i am using the correct port.